arduino/nina-fw #2: Error connecting to device

durasno, 2018-10-30:
Thanks for this branch!

When I follow the steps in the [README](https://github.com/arduino/nina-fw/blob/master/README.md) I get the following error with the final step ("Use esptool to flash the compiled firmware"):

Operating System: OS X Yosemite
Board: Arduino MKR WiFi 1010

command:
`esptool.py -p /dev/cu.usbmodem411 --trace flash_id`

error:
```esptool.py v2.5.1
Serial port /dev/cu.usbmodem411
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 Timed out waiting for packet header
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 Timed out waiting for packet header
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 Timed out waiting for packet header
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.102 Timed out waiting for packet header
.TRACE +1.812 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.102 Timed out waiting for packet header
_TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
[... 554 lines cut ...]
TRACE +0.104 Timed out waiting for packet header
_TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.102 Timed out waiting for packet header
_TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```
I also get a similar message when I try to flash firmware:
`esptool.py -p /dev/cu.usbmodem411 --trace write_flash 0x30000 ~/nina-fw-wpa2_enterprise/build/nina-fw.bin`

facchinm, 2018-10-31:
Hi  @durasno ,
did you flash on the SAMD21 the [SerialNINAPassthrough](https://github.com/arduino-libraries/WiFiNINA/tree/master/examples/Tools/SerialNINAPassthrough) sketch?
> Load the Tools -> SerialNINAPassthrough example sketch on to the board
Also ensure to remove the foam from the board and not to use pins 13 and 14 for anything else (since the passthrough uses that pins as a side channel).
Let me know if it works!

durasno, 2018-10-31:
Hi @facchinm,

Yes, I flashed it with the SerialNINAPassthrough sketch. The board is on breadboard with nothing connected except a LiPo battery and the USB cable.

facchinm, 2018-11-05:
@sandeepmistry any idea? Did you ever get something like this on OSX?

sandeepmistry, 2018-11-05:
Seems ok here with both the `/dev/cu.` and `/dev/tty.` interfaces.

I'm using esptool 2.5.0 which is part of IDF 3.1:

```
$ python /Users/smistry/git/esp-idf/components/esptool_py/esptool/esptool.py version
esptool.py v2.5.0
2.5.0
```

```
python /Users/smistry/git/esp-idf/components/esptool_py/esptool/esptool.py  -p /dev/cu.usbmodem14101 --trace flash_id
esptool.py v2.5.0
Serial port /dev/cu.usbmodem14101
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 Timed out waiting for packet header
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 6 bytes: 010804000755
TRACE +0.000 Read 1 bytes: 55
TRACE +0.000 Read 5 bytes: 5500000000
TRACE +0.000 Read 3 bytes: c0c001
TRACE +0.000 Received full packet: 010804000755555500000000
TRACE +0.001 Read 5 bytes: 0804000755
TRACE +0.001 Read 4 bytes: 55550000
TRACE +0.000 Read 7 bytes: 0000c0c0010804
TRACE +0.000 Received full packet: 010804000755555500000000
TRACE +0.001 Read 9 bytes: 000755555500000000
TRACE +0.000 Read 4 bytes: c0c00108
TRACE +0.000 Received full packet: 010804000755555500000000
TRACE +0.001 Read 6 bytes: 040007555555
TRACE +0.000 Read 3 bytes: 000000
TRACE +0.000 Read 2 bytes: 00c0
TRACE +0.000 Received full packet: 010804000755555500000000
TRACE +0.000 Read 4 bytes: c0010804
TRACE +0.000 Read 2 bytes: 0007
TRACE +0.000 Read 1 bytes: 55
TRACE +0.000 Read 1 bytes: 55
TRACE +0.000 Read 1 bytes: 55
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Received full packet: 010804000755555500000000
TRACE +0.000 Read 3 bytes: c00108
TRACE +0.000 Read 1 bytes: 04
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 07
TRACE +0.000 Read 2 bytes: 5555
TRACE +0.000 Read 1 bytes: 55
TRACE +0.000 Read 2 bytes: 0000
TRACE +0.000 Read 2 bytes: 0000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Received full packet: 010804000755555500000000
TRACE +0.000 Read 3 bytes: c00108
[... 1525 lines cut ...]
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Received full packet: 01090200000000000000
TRACE +0.000 command op=0x09 data len=16 wait_response=1 timeout=3.000 data=2420006000000070ffffffff00000000
TRACE +0.000 Write 26 bytes: 
    c000091000000000 0024200060000000 | .........$ .`...
    70ffffffff000000 00c0             | p.........
TRACE +0.003 Read 1 bytes: c0
TRACE +0.000 Read 2 bytes: 0109
TRACE +0.000 Read 1 bytes: 02
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 2 bytes: 0000
TRACE +0.000 Read 2 bytes: 0000
TRACE +0.000 Read 2 bytes: 00c0
TRACE +0.000 Received full packet: 01090200000000000000
Manufacturer: c2
Device: 2815
Detected flash size: 2MB
Hard resetting via RTS pin...
```

facchinm, 2018-11-16:
Moving to #6

closed 2018-11-16 by facchinm