arduino/nina-fw #4: Error while erasing flash memory

tigoe, 2018-11-16:
While using the WiFiNINA Firmware/Certificates updater, I get this error when I upload to a WiFi 1010, both when I try to update the firmware and when I try to upload certificates. The firmware update seems to have happened successfully, but the certificate upload not working.

Steps needed to reproduce:
* Upload FirmwareUpdater to WiFi 1010
* Open FirmwareUpdater tool
* Test connection to serial port, OK
* Click Upload Firmware, produces error  "Error while erasing flash memory"

Firmware version check:
* open WiFiNINA WiFiSSL client, add the following after line 47:
```
Serial.println(fv);
```
Result: 1.2.0

SSL Check:
* Change SECRET_PASS and SECRET_SSID to check if SSL working
Result:
```
Attempting to connect to SSID: myNetwork
Connected to wifi
SSID: myNetwork
IP Address: 192.168.0.19
signal strength (RSSI):-41 dBm

Starting connection to server...

disconnecting from server.
```

Same result whether using numeric or name IP
Changed server name to www.arduino.cc, same result

Checking certificate change:
* Upload firmwareUpdater sketch
* Open firmware/Certificate updater
* Add new domain, result is "Error while erasing flash memory"
* Go back to WiFiSSLClient, try again with www.arduino.cc Same negative result as above.
* try again with my own known domain. Same negative result as above.

tigoe, 2018-11-16:
Update: after moving Arduino app to Applications folder on MacOS 10.12.6, firmware uploading reports successful upload of certificates. However, WiFiSSL Client still fails, using www.arduino.cc or my own domain. 

Note that I changed the Host: line  to the appropriate host and the GET line to simply GET / HTTP/1.1. I also tried by making up a unique path ( GET /aadvark-snow-booties.html HTTP/1.1) and then searching the server and server error logs for that result. The request does not seem to be reaching the server at all, even though the device is successfully on the network.

facchinm, 2018-11-16:
Hi Tom,
would you mind trying to connect to https without uploading the website certificate? There's a bunch of factory provided certs that should work for most websites.
In the meantime I'll try to reproduce the  "Error while erasing flash memory"

tigoe, 2018-11-16:
Sure thing. New test:
* uploaded firmwareUpdater sketch
* updated firmware using firmware/cert uploader
* got "error verifying flashed firmware"
* repeated process. got "error while erasing flash memory"
* re-uploaded firmwareUpdater sketch. repeated process. firmware updated successfully.
* tried with arduino.cc. Got same response as above
* tried with my own domain. Got same response as above

Server logs on my server (checked http and https access logs and error logs) show no sign of my requests.

facchinm, 2018-11-16:
Closing with 35d4a3c for the second part of the test (now default certificates are correctly handled), moving to #2 since they both are about problems communicating with the NINA module using external UART.

closed 2018-11-16 by facchinm