arduino/nina-fw #6: Problems flashing firmware (esptool/WiFiNINA FW Update Plugin)

facchinm, 2018-11-16:
Some users are reporting difficulties in flashing the NINA firmware (with both `esptool` and the IDE plugin) (see #2, #4)

The flashing procedure is similar on both cases; a side channel (pins 13 and 14) are being used to route the NINA programming serial port with the SAMD one. 
GPIO0 and RESET (internal signals) are being used to trigger a reboot of the ESP32 core into "download mode".
In the `FW Updater Plugin` case, the communication is encapsulated in a specially crafted format, while with `esptool` the USB Serial is simply routed to the internal serial port.

ATM, both computers affected are running OSX.

gojimmypi, 2018-12-09:
+1 on problems flashing nina; Arduino IDE 1.8.9 Hourly Build 2018/12/06 03:33

In trying to upload the "CheckFirmwareVersion" code, often will cause my COM port to completely disappear from Windows! Even a power cycle of board and computer does not resolve. The only way I've found to fix is when uploading the WiFiNINAWiFiSSLClient, somehow during initialization of USB the COM port is "fixed". (?)

Here's the log when trying to upload the CheckFirmwareVersion via COM port:

```
Sketch uses 14128 bytes (5%) of program storage space. Maximum is 262144 bytes.
Forcing reset using 1200bps open/close on port COM3
PORTS {COM3, } / {} => {}
PORTS {} / {COM3, } => {COM3, }
Found upload port: COM3
C:\Users\gojimmypi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0/bossac.exe -i -d --port=COM3 -U true -i -e -w -v C:\Users\GOJIMM~1\AppData\Local\Temp\arduino_build_732552/CheckFirmwareVersion.ino.bin -R 
An error occurred while uploading the sketch
processing.app.SerialException: Error opening serial port 'COM3'.
	at processing.app.Serial.<init>(Serial.java:147)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$4.<init>(SerialMonitor.java:101)
	at processing.app.SerialMonitor.open(SerialMonitor.java:101)
	at processing.app.AbstractMonitor.resume(AbstractMonitor.java:104)
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2098)
	at processing.app.Editor.access$1400(Editor.java:113)
	at processing.app.Editor$UploadHandler.run(Editor.java:2076)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - COM3; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	at processing.app.Serial.<init>(Serial.java:136)
	... 8 more
Error opening serial port 'COM3'.
```

I'm able to otherwise upload code such as the blink demo via the serial port without problems.

I also tried uploading the CheckFirmwareVersion via the Atmel ICE as well; no joy.

facchinm, 2018-12-10:
@gojimmypi your problem is not really related with the issue (which is about updating the Nina firmware and not the arduino sketch). Anyway, which Windows version are you using? Did you remove the foam from the board and/or tried to upload without anything connected to the pins?

gojimmypi, 2018-12-11:
@facchinm thanks for your reply. Indeed you are correct: the problem I saw was not related to this issue. I was using Windows 10 1809 and experimenting with the Atmel ICE SWD to program and debug the MKR 1010 (yes, foam removed) with Atmel Studio, Apparently after sending code via the ICE, "weird things" happened with the bootloader causing the later serial upload problems (the COM port would go missing) 

I misunderstood the bootloader on the SAMD vs the Nina firmware on the ESP32.

I wrote up a blog about my experiences here:

[https://gojimmypi.blogspot.com/2018/12/swd-debugging-arduino-mkr-wifi-1010.html](https://gojimmypi.blogspot.com/2018/12/swd-debugging-arduino-mkr-wifi-1010.html)

I did later learn that flashing the ESP32 Nina is a completely different exercise:

[https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/Tools/SerialNINAPassthrough/SerialNINAPassthrough.ino](https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/Tools/SerialNINAPassthrough/SerialNINAPassthrough.ino)

closed 2019-03-26 by sandeepmistry