arduino/nina-fw #7: Cannot flash a new firmware to ESP32

dtltechnologies, 2018-12-17:
Hello,

I'm trying to load a freshly compiled nina-fw.bin file into an Arduino MKR 1010 board, with firmware version 1.1.0.

I followed the readme file and followed the steps. I run this command and get that error:

`$ python /home/Dorian/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/Dorian/esp/nina-fw-master/build/bootloader/bootloader.bin 0x30000 /home/Dorian/esp/nina-fw-master/build/nina-fw.bin 0x8000 /home/Dorian/esp/nina-fw-master/build/partitions.bin
esptool.py v2.5.0
Found 1 serial ports
Serial port COM6
Connecting...
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 24:0a:c4:ad:1f:54
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Compressed 13424 bytes to 8833...
Wrote 13424 bytes (8833 compressed) at 0x00001000 in 0.8 seconds (effective 134.7 kbit/s)...
Hash of data verified.
Compressed 806880 bytes to 493599...
Writing at 0x00030000... (3 %)
A fatal error occurred: Timed out waiting for packet header
`

It seems like the bootloader is successfully loaded, but the program part is not.

Any idea where the issue could be?

Thanks,

Dorian.

facchinm, 2018-12-17:
Hi @dtltechnologies ,
can you try running `make firmware` (which produces a "fat" binary called `NINA_W102.bin`) and flash it with `python /home/Dorian/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 NINA_W102.bin` ? 
If it stops at the same point it could be a problem related with our SerialPassthrough sketch.

dtltechnologies, 2018-12-17:
Hi,

This workaround works fine (I have not checked my program yet, but the firmware seems to be successfully uploaded):

```
$ python /home/Dorian/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 NINA_W102.bin
esptool.py v2.5.0
Found 1 serial ports
Serial port COM6
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 24:0a:c4:ad:1f:54
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Compressed 1003520 bytes to 537854...
Wrote 1003520 bytes (537854 compressed) at 0x00000000 in 48.0 seconds (effective 167.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Is there any difference between this "fat" version and the one split into 3 parts?

Thanks,

Dorian.

facchinm, 2018-12-17:
No real differences, but blank space is padded with 0xFF (and probably more important, the offset communication only happens once)

dtltechnologies, 2018-12-17:
Alright, everything works fine now, thank you.
It may be worth considering an update of the documentation?

facchinm, 2018-12-19:
Could be a good addition to the documentation! Would you mind adding a PR with the proposed changes? Thanks!

closed 2018-12-19 by facchinm

jwestmoreland, 2019-02-12:
Hello,
I don't see the Pull Request - can someone do that please?

Thanks,
John W.