arduino/nina-fw #10: SSL sites that can't connect

ladyada, 2019-02-11:
while trying various sites i've found a few that don't connect, not sure what the commonality is other than SSL - the root cert appears in the collection.

- https://randomfox.ca/floof

danlemire, 2019-03-31:
Is there anyway possible to enable accepting of self-signed certificates?

facchinm, 2019-04-02:
@danlemire it will be possible very soon :slightly_smiling_face:

nevercast, 2019-06-25:
That website is using CloudFlare for the SSL, so it could be an issue with alternate domains, since this is how CloudFlare reuses certificates. 

Or it could be the key exchange is new and unsupported by the implementation in ESP IDF

edwRoland, 2019-11-14:
I have still the same problems with connecting to a specific webside.
Others work but the website "www.bayrol-poolaccess.de" doesn't.

I've updated the SSL root certificates from the said website and its root certificate provider: digicert.com

Is there any way how i can debug this? Error Messages or Debug Messages to help me what I'm doing wrong?

Thank you very much for your help and time!

Arduino 1.8.10
Arduino NANO 33 IoT
NINA firmware version 1.2.4 Arduino MKR WiFi 1010

bricemiramont, 2020-05-06:
Hi, same here. I'd like to have my own pki as my arduino 33 ioT will be offline and I need secure communication with another device using wifi and https. I have a custom root CA, a cert (signed with my root CA) on my web server but the arduino does not get the root CA to check the web server cert when I try to load it with firmwareupdater.
Will it be possible to use self-signed certificates ?

VEGA-Processor, 2023-12-06:
Hi @ladyada @bricemiramont  I was trying WiFiSSLClient example in WiFiNiNA library, but it is not working. (All other examples are working)

Arduino IDE: 1.8.19
Board: ARIES IoT with NiNA-W102
Firmware Version: 1.5.0 (latest)

Please help me with this. Thank You.