arduino/nina-fw #11: Build Error

jwestmoreland, 2019-02-12:
Hello,

I'm having the following issue making a build:
CXX build/arduino/libraries/WiFi/src/WiFiUdp.o
CXX build/arduino/libraries/WiFi/src/WiFi.o
In file included from C:/esp/nina-fw/arduino/libraries/WiFi/src/WiFi.cpp:25:0:
C:/esp/esp-idf/components/lwip/include_compat/apps/sntp/sntp.h:2:2: warning: #warning "This header file is deprecated, please include lwip/apps/sntp.h instead." [-Wcpp]
 #warning "This header file is deprecated, please include lwip/apps/sntp.h instead."
  ^
C:/esp/nina-fw/arduino/libraries/WiFi/src/WiFi.cpp: In member function 'int WiFiClass::ping(uint32_t, uint8_t)':
C:/esp/nina-fw/arduino/libraries/WiFi/src/WiFi.cpp:114:54: error: 'inet_addr_from_ipaddr' was not declared in this scope
   inet_addr_from_ipaddr(&to.sin_addr, ip_2_ip4(&addr));
                                                      ^
make[1]: *** [/c/esp/esp-idf/make/component_wrapper.mk:292: libraries/WiFi/src/WiFi.o] Error 1
make: *** [/c/esp/esp-idf//make/project.mk:527: component-arduino-build] Error 2

This is obviously related to the ping command - I tried this on a 'clean' machine to verify - got same exact failure that I had on 1st build machine.

Thanks In Advance,
John W.

jwestmoreland, 2019-02-12:
Also - with the workaround posted on this site - I got this:

john@pc MINGW32 /c/esp/nina-fw
# python /c/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM11 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 NINA_W102.bin
esptool.py v2.6
Serial port COM11
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 80:7d:3a:86:05:d8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Compressed 1050624 bytes to 567834...
Writing at 0x00000000... (2 %)Traceback (most recent call last):
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2959, in <module>
    _main()
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2952, in _main
    main()
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2698, in main
    operation_func(esp, args)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2158, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio * 2)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 632, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 354, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 322, in command
    self.write(pkt)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 285, in write
    self._port.write(buf)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 323, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout

facchinm, 2019-02-12:
Please checkout v3.1 of esp-idf before building and flashing (we are building against it so problems are expected when building against  any other version.

jwestmoreland, 2019-02-12:
Oops - see that now.
OK - doing that - will report back.
Thanks,
John

jwestmoreland, 2019-02-12:
ok - under 3.1 - builds OK - 
still getting this though:
john@WEI-WS2 MINGW32 /c/esp/nina-fw
$ python /c/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM17 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 NINA_W102.bin
esptool.py v2.5.0
Serial port COM17
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 84:0d:8e:34:62:20
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Compressed 1001472 bytes to 538027...
Writing at 0x0003c000... (48 %)Traceback (most recent call last):
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2849, in <module>
    _main()
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2842, in _main
    main()
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2594, in main
    operation_func(esp, args)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2074, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 95, in inner
    return func(*args, **kwargs)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 616, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 338, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 306, in command
    self.write(pkt)
  File "C:/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 269, in write
    self._port.write(buf)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 323, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout

facchinm, 2019-02-12:
@jwestmoreland did you remove the foam from the board (since the fw update is pushed through external pin headers) ?

jwestmoreland, 2019-02-12:
Yes, of course.  You can't even download a sketch if the boards are in that foam.

facchinm, 2019-02-12:
To sort everything out, I'd ask you to test the IDE WiFi Updater plugin too.
To use it, do the following:
- update IDE to 1.8.8 or newer
- copy NINA_W102.bin to `$ide_base_folder/tools/WiFi101/tool/firmwares/NINA/1.2.1/` replacing the existing one
- open the plugin (from Tools menu, WiFi101 / WiFiNINA Firmware Updater) 
- flash FirmwareUpdater from WiFiNINA examples
- try updating using the plugin

Thanks!

jwestmoreland, 2019-02-13:
That worked - but the build doesn't seem to work - never get passed:
Attempting to connect to SSID: xxxx

I put the old build back, did again, and tried with the WiFi web client  (sketch) and it connected ok.

So, the new build downloads using the plugin - but doesn't seem to work.

So, can we gather from this that the esptool.py method isn't really working - maybe something needs to be fixed with the sketch for that - what about changing timeouts - is that possible?

Thanks.
(So far, I just did this on the MKR 1010 - I can do for Vidor also just I can't get the plugin to work with 1.2.1 (yet).)

facchinm, 2019-02-13:
Mmmmh if it arrives at `Attempting to connect` phase it means that the firmware has been flashed correctly at least. This patch https://github.com/arduino/nina-fw/commit/ee167aba00534a10608c7a106f19db9f4ad7229b may change the antenna sensitivity quite a bit, so maybe try to reflash after reverting it.
On Vidor, we still didn't validate the new firmware so it's excluded from the dropdown list (but you can safely move the new firmware to 1.1.0 folder and flash it :wink: )

jwestmoreland, 2019-02-13:
ok - I will try that - just got this BTW with the 1010 - first time I saw it get this far:
John@WEI-W530 MINGW32 /c/esp/nina-fw
$ python /c/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM20 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 NINA_W102.bin
esptool.py v2.6
Serial port COM20
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 80:7d:3a:85:e1:9c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Compressed 949248 bytes to 503796...
Writing at 0x00078000... (100 %)
A fatal error occurred: Timed out waiting for packet header

jwestmoreland, 2019-02-13:
That worked - this is from the 4K:
WiFiNINA firmware check.

Firmware version installed: 1.2.1
Latest firmware version available : 1.2.1

Check result: PASSED

closed 2019-03-26 by sandeepmistry