arduino/nina-fw #18: UDP stops working on heavy load

javier-paris, 2019-04-12:
I'm trying to receive udp packets sent with a frequency of around 100Hz on a MKR 1010. The Wifi module stops working after a while. Increasing the frequency seems to make the module fail earlier.

The MKR is just counting the number of packets received:

```
#include <SPI.h>
#include <WiFiNINA.h>
#include <WiFiUdp.h>

WiFiUDP udp;

char SSID[]="...";
char pass[]="...";
int local_port = 4242;

int pkts_rcvd=0;

void setup() {
  Serial.begin(9600); while(!Serial);
  
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.write("No WiFi module detected\n");
    while (true);
  }

  Serial.print((String)"Attempting connection to "+SSID+"... ");
  while (WiFi.begin(SSID, pass)!= WL_CONNECTED) delay(1000);
  Serial.println("Success!");
  
  if(!udp.begin(local_port)) {
    Serial.println("Could not open socket");
    while(true); 
  }
}

void loop() {
  int pkt_len;
  
  pkt_len = udp.parsePacket();

  if(pkt_len>0) {
    char pkt[pkt_len];
    udp.read((char *) pkt, pkt_len);
    Serial.println(pkts_rcvd);
    pkts_rcvd++;
  } 
}
```

And I'm sending the udp packets from a Linux box using the following program:

```
#include <stdlib.h>
#include <unistd.h>
#include <stdio.h>
#include <sys/types.h>
#include <sys/socket.h>
#include <netinet/in.h>
#include <arpa/inet.h>

int main() {
	int s;
	struct sockaddr_in dst;
	char data[512];
	
	s=socket(PF_INET, SOCK_DGRAM, 0);

	if(s==-1) exit(-1);

	dst.sin_family      = AF_INET;
	dst.sin_port        = htons(4242);
	dst.sin_addr.s_addr = inet_addr("...");

	while(1) {
		usleep(2000); // 2ms => 500Hz

		if(sendto(s, data, 512, 0, (struct sockaddr *) &dst, sizeof(dst))==-1) {
			perror("Error sending data: ");
		} 
	}
}
```

After a while parsePacket always returns 0. Looking at the network traffic with tcpdump, it seems to happen when the 1010 sends a gratuitous arp:

```
23:37:59.608435 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512  <== This are normal packets
23:37:59.610550 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:37:59.612663 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:37:59.614777 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:37:59.616891 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:37:59.617279 ARP, Request who-has arduino-7c0c tell 0.0.0.0, length 28 <= Here the 1010 sends a gratuitous arp
23:37:59.619011 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:37:59.619159 ARP, Request who-has arduino-7c0c tell arduino-7c0c, length 28
23:37:59.621101 ARP, Request who-has 192.168.1.1 tell arduino-7c0c, length 28
23:37:59.621131 ARP, Reply 192.168.1.1 is-at b8:27:eb:a0:b6:06 (oui Unknown), length 28
23:37:59.621148 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:37:59.623235 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:37:59.623369 IP arduino-7c0c > 192.168.1.1: ICMP arduino-7c0c udp port 4242 unreachable, length 36 <= Now the 1010 is saying that the port is unreachable
23:37:59.625346 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512

```

After a short while the 1010 ignores arp requests.

```
23:38:02.593061 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:38:02.595168 IP 192.168.1.1.44643 > arduino-7c0c.4242: UDP, length 512
23:38:03.718631 ARP, Request who-has arduino-7c0c tell 192.168.1.1, length 28
23:38:04.758624 ARP, Request who-has arduino-7c0c tell 192.168.1.1, length 28
23:38:05.798625 ARP, Request who-has arduino-7c0c tell 192.168.1.1, length 28
23:38:06.940159 ARP, Request who-has arduino-7c0c tell 192.168.1.1, length 28
```

sandeepmistry, 2019-04-26:
@javier-paris thanks for the detailed issue report! I've able to reproduce this locally.

One note, if I add a small delay to the sketch, it seems to avoid the crash so far ...

```
#include <SPI.h>
#include <WiFiNINA.h>
#include <WiFiUdp.h>

WiFiUDP udp;

char SSID[] = "";
char pass[] = "";
int local_port = 4242;

int pkts_rcvd = 0;

void setup() {
  Serial.begin(9600); while (!Serial);

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.write("No WiFi module detected\n");
    while (true);
  }

  Serial.print((String)"Attempting connection to " + SSID + "... ");
  while (WiFi.begin(SSID, pass) != WL_CONNECTED) delay(1000);
  Serial.println("Success!");

  IPAddress ip = WiFi.localIP();

  Serial.println(ip);

  if (!udp.begin(local_port)) {
    Serial.println("Could not open socket");
    while (true);
  }
}

void loop() {
  int pkt_len;

  pkt_len = udp.parsePacket();

  if (pkt_len > 0) {
    char pkt[pkt_len];
    udp.read((char *) pkt, pkt_len);

    Serial.println(pkts_rcvd);
    pkts_rcvd++;
  }

  delay(10); /// <--- added
}
```

sandeepmistry, 2019-04-26:
Hi @javier-paris,

I've made a pull request with a binary you can try, https://github.com/arduino/nina-fw/pull/20#issuecomment-487142506

I'm not sure if it's the best fix for the issue, but I don't see a crash when I try it locally with the steps to reproduce you've provided.

javier-paris, 2019-04-30:
Hi @sandeepmistry , 

I've tried the new binary and it's working.

Thanks!

closed 2019-04-30 by javier-paris

sandeepmistry, 2019-05-01:
@javier-paris great, thanks for testing!

I'll see if we can get #20 merged for the next release.