arduino/nina-fw #19: [Enhancement] Add a digitalRead SPI command

anecdata, 2019-04-25:
nina-fw currently supports pinMode, analogWrite, and DigitalWrite. It would be very useful to also be able to read from the ESP32's digital pins via SPI from an attached processor.

Virtually infinite applications, but one in particular would be to use input(s) on the ESP32 as an address pin or identifier, so that the host processor can determine the form factor (and therefore the exposed pins and capabilities) of the ESP32 co-processor.

xelll, 2019-04-28:
it's also useful to be able to read anlog pins value

anecdata, 2019-05-15:
Some use cases:
- GPIO expansion;
- Adafruit ESP32 Breakout and Adafruit ESP32 Feather have battery voltage on (unexposed) pin A13 / I35;
- Ability to read specially-configured ESP32 co-processor inputs would allow the main processor to determine the form factor of the attached ESP32 co-processor (e.g., Adafruit Airlift Breakout vs. Adafruit ESP32 Feather vs. Adafruit ESP32 Breakout, etc.), and adapt to its unique capabilities.

sandeepmistry, 2019-09-18:
Hi @anecdata,

At this time, we are not interested in implementing this feature, I'm marking the issue as won't fix.

closed 2019-09-18 by sandeepmistry

anecdata, 2021-06-28:
Now implemented as part of #67

xelll, 2021-07-13:
> Now implemented as part of #67

:+1: