arduino/nina-fw #30: Arduino wifi rev 2 - WiFi web server stop responding

awsiukie, 2019-07-22:
Hey,

I think there is a BUG in WiFININA firmware. I have tested 1.2.1 and 1.2.3 firmware. I have almost code from example library extended with few items. Issue occur around every 15mins and WiFi server stops responding, meaning it's no longer possible to connect to port 80 with web browser, however you still can ping the device.  I have described issue here:
https://forum.arduino.cc/index.php?topic=626907.0
but so far no luck. 
I've noted 2 type of behaviors:
1. after a while, usually 10-15mins, server status return 0, but it's not possible to start it with begin. It is still in status 0.
2. after a while, usually 10-15mins, server return status 1 but it's not responding. 
to fix this issue, I need to restart arduino, either software wise or hardware button.

I've been trying to flood it with Web requests but it was still working. It's somewhat connected with time and what happens in mean time. One thing is sure, more connections, more chance to hang it up earlier. 

PS Would be nice to have some close and reopen socket and some debug functions to see for instance on serial whether there are some connections hanging (like netstat).

Rocketct, 2019-07-25:
Hi @awsiukie i have read the post to check the issue a need a complete sketch to reproduce the issue, could you share a minimal one?

awsiukie, 2019-07-26:
Hey,

This setup is really simple. Just connect HC-12 to arduino:
*  GND to GND
* Vcc to 5V
* HC12 ping TX --> D5 (PWM)
* HC12 ping RX --> D6 (PWM)
there is nothing else there except HC-12. I'm sending data every 2s from the other arduino.

[receiver.zip](https://github.com/arduino/nina-fw/files/3434679/receiver.zip)

Rocketct, 2019-07-26:
Hi @awsiukie tested with the following sketch is a simplified version of your i have removed the HC-12, and tested only the connectivity, with the firmware version: 1.2.4 , i have run for two hours and each time that from the browser i have update the page to his ip it had response with:

```
12:11:34.781 -> GET / HTTP/1.1
12:11:34.781 -> Host: "myip"
12:11:34.814 -> User-Agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:68.0) Gecko/20100101 Firefox/68.0
12:11:34.914 -> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
12:11:34.980 -> Accept-Language: en-US,en;q=0.5
12:11:35.013 -> Accept-Encoding: gzip, deflate
12:11:35.046 -> Connection: keep-alive
12:11:35.080 -> Upgrade-Insecure-Requests: 1
12:11:35.113 -> Cache-Control: max-age=0
12:11:35.113 -> 
12:11:35.146 -> client disconnected
``` 
here the modified sketch, i have removed the part of thre HC-12, i don't have one, the serial 1 communication and lasta i have moved the client.print() at the end of the packet with his sketch and with the firmware it works fine, i have also get a wireshark to check if the board send the response to the request and it show the packet sent by the board, this is how i have change the sketch for my test, 
```c++
//#include <SoftwareSerial.h>
#include <SPI.h>
#include <WiFiNINA.h>
#include "arduino_secrets.h"
#include "foliownik.h"

//SoftwareSerial HC12(5, 6); // HC-12 TX Pin, HC-12 RX Pin

//dane pogodowe z czujników
frame received_frame;

int last_data = 0;
unsigned long int klient = 0;

char ssid[] = SECRET_SSID;        // your network SSID (name)
char pass[] = SECRET_PASS;        // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                 // your network key Index number (needed only for WEP)
int status = WL_IDLE_STATUS;
WiFiServer server(80);            // Serwer będzie nasłuchiwał na porcie 80

unsigned long startMillis;
unsigned long CurrentMillis;
unsigned long reset_counter;
unsigned long reset_call = 510000;

void setup() {

  Serial.begin(9600);
  //Serial1.begin(9600);
//  HC12.begin(9600);          

  analogWrite(A5, 0);         

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);

    status = WiFi.begin(ssid, pass);
    delay(4000);
  }

  server.begin();
  printWifiStatus();
  startMillis = millis();
  reset_counter = startMillis;

//  Klient = 0;
}
[... 96 lines cut ...]
    Serial.println("client disconnected");

  }

}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
```

first of all are you ever using the 1.2.1 and 1.2.3? have you try to update to the 1.2.4? after would you try the server example and check if after 15 minutes stop to works too?

awsiukie, 2019-07-27:
Hey @Rocketct, 

thanks for looking at it.
In general if there are no clients it works for long time it do not hangs. When i get like 3-4 connections per minute after like 15-20 minutes it stops working. It won't work longer that 30minutes but there must be like 3-4 connections a minute over all time. I'm running zabbix to get the data. Here is diagram how it looks like:
[image]
These blank spaces is when i do reset. With counter that is right now implemented, getting data looks like this:
[image]
After like 4 days running it hanged again. Now need to get there and do hard reset. Such hangs are more rare with shorter reset timer. If that is around 15minutes then it hangs often and if that's every 30 mins then reset function actually does not work. 
Remotely i see this when i run wireshark:
[image]

I need to connect to serial1 and see what is dropping there. I need to write more debug stuff into serial1. 

I didn't try running basic sketch for 15 minutes w/o HC12. I can try. I need to run really basic code, but it is really strange why it hangs. What only comes to my mind is the buffer overflow or something. Or i made bubu in the code...

Rocketct, 2019-07-29:
@awsiukie thanks for reply, ok i'll try to set the tests in order to see if your issue is replicated, please align the nina's firmware to last version, i'll wait for your report.

awsiukie, 2019-07-30:
Hey @Rocketct 

Made some code updates. I've solved problem with transmission that i had. I've created larger frame_buf (incoming HC12 buffer) to 500B from sizeof(frame), added 50ms delay between HC12 byte read to fix transmission issue. Now i don't get CRC errors that i had while getting data from air over HC12.

[odbiornik_v2.2.zip](https://github.com/arduino/nina-fw/files/3445651/odbiornik_v2.2.zip)

I also updated firmware to 1.2.4 from hourly build. I have feeling that it works worse than 1.2.3 as it started to hang after like 5mins, I'm not sure if that is coincidence or what, but went back to 1.2.3.  With new code on 1.2.3 firmware got times between hangs:
* 2.5hrs
* 45mins

BR
Lukas

Rocketct, 2019-09-06:
@awsiukie coyld you make some test without the htC12? seems really strange that 1.2.4 behavior, i'll investigate further, thanks for the update

sandeepmistry, 2019-09-18:
@Rocketct do you have any updates on this?

awsiukie, 2019-09-18:
> 
> 
> @awsiukie coyld you make some test without the htC12? seems really strange that 1.2.4 behavior, i'll investigate further, thanks for the update

I  need to build some other project to test it out. I replaced that arduino with different approach. I moved to arduino micro and ESP01 as a client. I don't use WiFi one as a server due to this issue. 

Once I will find some more time will do test with pure and empty arduino WiFi. HC12 was only thing connected to it to get data over long distance.

BrianG4, 2020-07-29:
I have been experiencing a similar problem:

WifiServer on port 80 stops working after a while.

Factors in my testing seem to be:
- more than 1 client connected to wifiAP
- browser (seems to work ok with many browser/os combos had problem with Chrome on a particular version of android (not others...) - made better with pre-loading pages turned off)

I thought it may be caused by some tcp connections being left un-closed.  I'm not sure, but it seems as though disconnecting a device from the wifi, while another is still connected, does not fire Wifi.onDisconnect(), which in turn does not call CommandHandlerClass::onWiFiDisconnect(), which does not call CommandHandlerClass::handleWiFiDisconnect().

I modified CommandHandler.cpp (nina-fw-master 1.4.0) to copy the code from handleWiFiDisconnect() to startClientTcp(), in order to make sure unused connections were closed before opening new ones.

Inserted at Line 553/4, before connecting sockets:
```
  for (int i = 0; i < CONFIG_LWIP_MAX_SOCKETS; i++) {
    struct sockaddr_in addr; 
    size_t addrLen = sizeof(addr);
    int socket = LWIP_SOCKET_OFFSET + i;

    if (lwip_getsockname(socket, (sockaddr*)&addr, &addrLen) < 0) {
      continue;
    }

    if (addr.sin_addr.s_addr != 0) {
      // non-listening socket, close
      close(socket);
    }
  }
```
It has improved stability for me, but is a crude solution. Hope it helps.