arduino/nina-fw #36: Question about Uno WiFi OTA (remote UPDI) status

jandrassy, 2019-09-01:
In schematics of Uno WiFi I see that UPDI and reset of the 4809 are connected to ESP32. So OTA could be implemented in nina-fw acting as UPDI programmer.

Was this planed, attempted and failed for some reason or is it something postponed? 

I am thinking of making this if no one works on it or attempted it and failed for some objective reason.

facchinm, 2019-09-09:
Hi @jandrassy ,
the plan is to use a bootloader (https://github.com/arduino/ArduinoCore-megaavr/tree/master/bootloaders and https://github.com/arduino/nina-fw/pull/32 ) because the UPDI procedure was not stable enough to be production ready.
Anyway we have an almost pure arduino porting of an UPDI programmer as [muxTO](https://github.com/arduino/ArduinoCore-megaavr/tree/master/firmwares/MuxTO) (D11 converter you can find in the Nano Every) so porting it on the nina could be a very nice community project :wink:

sandeepmistry, 2019-09-18:
@jandrassy is your question answered? If so can we close this issue?

jandrassy, 2019-09-18:
yes. thank you

closed 2019-09-18 by jandrassy