arduino/nina-fw #37: Is it possible to force 802.11g?

andrea-m1, 2019-09-16:
Hi, I am troubleshooting a problem with 802.11n. Is it possible to force the module to use 802.11g only somehow?

sandeepmistry, 2019-09-18:
Hi @andrea-m1,

This feature is not supported at this time. I've marked it as an enhancement.

If you'd like to propose a pull request in the mean time that would be great!

tomazas, 2020-07-15:
Protocol selection could be implemented in "arduino\libraries\WiFi\src\WiFi.cpp" with additional call to **esp_wifi_set_protocol**, e.g.:
`uint8_t band = WIFI_PROTOCOL_11B|WIFI_PROTOCOL_11G|WIFI_PROTOCOL_11N;`
`esp_wifi_set_protocol(_interface, band);`

before the `esp_wifi_start();` is invoked.

**However, currently it is not possible to force ESP-IDF to use G specific protocol.** 
Only three options 802.11b, 802.11b/g or 802.11b/g/n are supported and neither of them is pure 802.11g.

See official code docs [here](https://github.com/espressif/esp-idf/blob/99d5959d8a319c26b5b907d26a4a5993a2775594/components/esp32/include/esp_wifi.h#L411) for proof.

---
Based on the issue count of [espressif](https://github.com/espressif/esp-idf/issues) the bad news is that you should not expect this feature to be implemented anytime soon.
Furthermore if your AP works only with higher (>802.11b) data rates the ESP-IDF will still try to send [some low-rate junk](https://esp32.com/viewtopic.php?f=14&t=4080) (of 802.11b protocol) over the air.

---
Offtopic - I'm just suprised how _boldly_ and _irresponsibly_ **u-blox** selected the cheapest so-incomplete espressif WiFi solution for the NINA devices.