arduino/nina-fw #48: DataTcp methods are prone to concurrency issues

cansik, 2020-06-23:
There are a lot of issues about Arduino UNO WiFi Rev2 bugs that occur from time to time, stopping the device from responding or leading to wrong packet data / sizes:

* https://github.com/arduino-libraries/WiFiNINA/issues/108
* https://github.com/arduino-libraries/WiFiNINA/issues/117
* https://github.com/hideakitai/ArduinoOSC/issues/16
* https://github.com/arduino-libraries/WiFiNINA/issues/106

It seems that this have to do with the SPI connection between the main μController (ATmega4809) and it's co-processor (ESP32) which is running the nina-firmware. To be more precise, I guess it has to do with the methods `availDataTcp` and `getDataTcp`.

The problem lays in their check-then-act calling, where someone first checks if data is available (and how much), and then requesting the actual data. Somehow, if there are a lot of new packages coming in, nina-fw struggles to fill them into the buffer at the right location. This then results into corrupted data and crashes a lot of higher level protocol implementations. Also, the ESP32 sometimes stop to work, maybe due an unwanted buffer overwrite. I've also tried to visualise it real quick for a better understanding.

[image]

Of course, you would like to try that yourself. For that I have created a minimal example, that shows the issue. Set the WiFi password, IP address and run the arduino code on an Arduino-UNO-Wifi-Rev2 and the Processing sketch on any computer (no additional dependencies are required except `WiFiNINA`):

- [Example Gist](https://gist.github.com/cansik/2f135a04d833f4be940f3cd385d14c75)
  - [receiver.ino](https://gist.github.com/cansik/2f135a04d833f4be940f3cd385d14c75#file-receiver-ino)
  - [sender.pde](https://gist.github.com/cansik/2f135a04d833f4be940f3cd385d14c75#file-sender-pde)

I am running this code on the most-recent WiFiNINA version and the most recent NINA-FW (1.3.0). This error happens most if two different messages are sent, because with only one, you will not see the overwrite. Here is a cropped example output. You even see the wrong overwriting of the buffer the longer it takes. This happens really quick if the timing of the packages is quite high (about 60 packages per second). Otherwise it happens too, but takes longer to occur (as every concurrency issue).

```
18:16:33.944 -> 1 received [10]
18:16:33.944 -> 2 received [14]
18:16:33.978 -> 1 received [10]
18:16:34.014 -> 2 received [14]
18:16:34.014 -> 1 received [10]
18:16:34.052 -> 2 received [14]
18:16:34.052 -> 1 received [10]
18:16:34.052 ->  received [10]
18:16:34.089 -> BUFFER: 6F 72 73 65 6C 6F 2F 64 6F 67 6F 72 73 65 0 0 0 0 0 0 0 0 0 0 
18:16:34.160 -> BUFFER: o r s e l o / d o g o r s e                     
18:16:34.227 ->  received [10]
18:16:34.227 -> BUFFER: 2F 64 6F 67 6C 6F 2F 64 6F 67 6F 72 73 65 0 0 0 0 0 0 0 0 0 0 
18:16:34.298 -> BUFFER: / d o g l o / d o g o r s e                     
18:16:34.371 ->  received [10]
18:16:34.371 -> BUFFER: 2F 64 6F 67 6F 72 73 65 6F 67 6F 72 73 65 0 0 0 0 0 0 0 0 0 0 
18:16:34.480 -> BUFFER: / d o g o r s e o g o r s e                     
18:16:34.514 ->  received [14]
18:16:34.552 -> BUFFER: 65 6C 6C 6F 2F 64 6F 67 6F 72 73 65 73 65 0 0 0 0 0 0 0 0 0 0 
18:16:34.622 -> BUFFER: e l l o / d o g o r s e s e                     
18:16:34.657 ->  received [10]
18:16:34.693 -> BUFFER: 6F 6F 64 62 79 65 2F 68 6F 72 73 65 73 65 0 0 0 0 0 0 0 0 0 0 
18:16:34.769 -> BUFFER: o o d b y e / h o r s e s e                     
18:16:34.843 ->  received [10]
18:16:34.843 -> BUFFER: 73 65 64 62 79 65 2F 68 6F 72 73 65 73 65 0 0 0 0 0 0 0 0 0 0 
18:16:34.918 -> BUFFER: s e d b y e / h o r s e s e                     
18:16:34.991 ->  received [14]
18:16:34.991 -> BUFFER: 6F 67 64 62 79 65 2F 68 6F 72 73 65 73 65 0 0 0 0 0 0 0 0 0 0 
18:16:35.061 -> BUFFER: o g d b y e / h o r s e s e       
```

To be honest, I did not yet look at the NINA-FW code, so I can only guess why it happens. But it seems that the receive buffer storing the incoming packages could be too small and overflow. Or there is something wrong with the logic there (are the incoming packages queued?) or is it a problem with the socket implementation of the Espressif firmware?

Would be great if someone more into this code could give me a hint where to dig deeper.

cansik, 2020-06-26:
It seems that WiFiServer (TCP Server) also has a bug. After sending some (not very fast) packages to the server, `client.available()` will return `64` three times, but `client.read()` contains only `0x00`. After that everything runs as normal. Could be related, but does not have to.

giulcioffi, 2020-07-02:
Hi @cansik, thank you for providing your sketches! They have been very useful for testing. 
The problem you are referring to was due to the fact that in the nina-fw the function to parse a new UDP packet was called in 2 places: 
- by the thread gpio0Updater() running continuously
- by the function that checks for available bytes: in case the number of available bytes was 0, the fw tried to parse another packet.
Then, sometimes it happened that the message stored in the buffer and the information about the length of the message went out of sync because of this. This means that sometimes you could have these errors:
- packetSize = 10, but BUFFER = /goodbye/h
- packetSize = 14, but BUFFER = /hello/dogorse
Once one of these 2 possible errors happens, from that moment on, the pointer to the right position of the messages in the buffer becomes totally misaligned, leading to a series fo erroneous messages.

Here is my PR: https://github.com/arduino/nina-fw/pull/49 
In order to update the fw, you can use the Firmare Updater that I linked in the PR comment, following the instructions in the [README](https://github.com/arduino/FirmwareUpdater).
I hope this solved your problem :)

closed 2020-07-09 by aentinger