arduino/nina-fw #52: 1.4.0 for the Arduino Uno Wifi Rev 2

imle, 2020-08-01:
I have spent some time trying to find info on this but it doesn't seem to be documented anywhere. I cannot get the wifi-rev-2 to upgrade its nina firmware to 1.4.0 through either the updater tool in the Arduino IDE or using the esptool and the latest firmware. Any idea what could be going on? The esptool finishes with what looks like a success but the status in code returns WL_NO_MODULE.

Joniras, 2020-11-25:
This may be the worst wy to do it but i did the following (and it worked for me, **No guarantee given ofc**): 
 - Downloaded the [latest release](https://github.com/arduino/nina-fw/tags) (.bin files) of nina-fw
        -  place the bin files in a folder called X.X.X (Version #) in the install folder ([Arduino Install folder]/tools/Wifi101/tool/firmwares)
 - Downloaded the latest release of [recaf](https://github.com/Col-E/Recaf/tags) (java bytecode editor) 
 - Open up Recaf and open the .jar file located under ([Arduino Install folder]/tools/Wifi101/tool)
 - Edit the Constructor of UpdateImpl to add another entry in the firmware-select box, for me this was `UpdaterImpl.fwAvailable.add(new NinaFlasher("NINA firmware", "1.X.Y", "firmwares/NINA/1.X.Y/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(Arrays.asList("Arduino Uno WiFi Rev2"))));`
- Export the Program to the same Location (Ctrl+S does nothing)
- Fire up Arduino again and update the firmware, check with 'CheckFirmwareVersion' if it worked

This worked for me but may not for you (due to newer versions, i tried with 1.4.1).

Aubrey4485, 2020-11-26:
Another less intensive approach...if you would like

Download the hourly (nightly) version of the Arduino IDE, i believe it is at 1.8.14 currently... the NINA firmware updater tool should now see the latest version available

Joniras, 2020-11-26:
Oh wow. I should have checked that first. 
Didn't think of nightly versions, thanks!!

closed 2025-03-20 by pennam