arduino/nina-fw #59: Can't find Firmware 1.4.2

fmatray, 2020-11-29:
Hello,
I installed the last version of WIFININA. 1.8.0. 
This library is suppose to work with the firmware 1.4.2.

I can't find the version 1.4.2 of the driver for the MKR 1010.
I tried the **nightly** Arduino IDE (last version is from september) on arduino.cc.
I also tried FirmwareUploader 0.1.2 found on GitHub, as suggested in an issue.
The only version I can find is 1.4.1.

How to find the version 1.4.2? 

Thanks a lot.

ocrdu, 2020-11-30:
See https://github.com/arduino-libraries/WiFiNINA/issues/140

p-brane, 2021-01-19:
I can't find version 1.4.2 either. The firmware is missing from the Arduino IDE hourly build from Dec 15, 2020 too. I up date a MKR WiFi 1010 to v 1.4.1 but I can only update the MKR Vidor 4000 to firmware version 1.1. It seems like you should be update it to v1.4.1 and above.

The Arduino - ScheduledWiFiSSLWebClient tutorial fails to connect because client. connect(sever,443) always returns 0 and is not a specified result in the library definition using WiFiNINA v1.7.1. I'd like to use the latest WiFiNINA library (v1.8.1) to see if it fixes the client.connect bug but need the latest firmware to test it.

per1234, 2021-01-20:
Regarding the original issue
>I can't find the version 1.4.2
...
The only version I can find is 1.4.1.

As is explained here: https://github.com/arduino-libraries/WiFiNINA/issues/148#issuecomment-757633761 the latest version expected by the WiFiNINA library was set to 1.4.2 in expectation of an imminent release of the firmware, but that release was delayed so the latest version truly is 1.4.1 and there is no 1.4.2. That spurious information has been corrected by https://github.com/arduino-libraries/WiFiNINA/pull/149 so you shouldn't get it if you use the latest release of WiFiNINA library.

The MKR Vidor 4000 firmware version is a separate issue. You can see here that the the firmware updater tool is configured to only provide up to the 1.1.1 version for that board:
https://github.com/arduino/WiFi101-FirmwareUpdater-Plugin/blob/659735277fd8c10a474782b63e1d999d681d3bbf/src/cc/arduino/plugins/wifi101/UpdaterImpl.java#L86
```java
fwAvailable.add(new WINCFlasher("WINC1501 Model B", "19.6.1", "firmwares/WINC1500/19.6.1/m2m_aio_3a0.bin", true, 1000000, new ArrayList<String>(asList("Arduino/Genuino MKR1000"))));
fwAvailable.add(new WINCFlasher("WINC1501 Model B", "19.5.4", "firmwares/WINC1500/19.5.4/m2m_aio_3a0.bin", true, 1000000, new ArrayList<String>(asList("Arduino/Genuino MKR1000"))));
fwAvailable.add(new WINCFlasher("WINC1501 Model B", "19.5.2", "firmwares/WINC1500/19.5.2/m2m_aio_3a0.bin", true, 1000000, new ArrayList<String>(asList("Arduino/Genuino MKR1000"))));
fwAvailable.add(new WINCFlasher("WINC1501 Model B", "19.4.4", "firmwares/WINC1500/19.4.4/m2m_aio_3a0.bin", true, 1000000, new ArrayList<String>(asList("Arduino/Genuino MKR1000"))));
fwAvailable.add(new WINCFlasher("WINC1501 Model A", "19.4.4", "firmwares/WINC1500/19.4.4/m2m_aio_2b0.bin", true, 115200, new ArrayList<String>(asList("Arduino WiFi 101 Shield"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.4.1", "firmwares/NINA/1.4.1/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010", "Arduino NANO 33 IoT"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.4.1", "firmwares/NINA/1.4.1/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(asList("Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.4.0", "firmwares/NINA/1.4.0/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010", "Arduino NANO 33 IoT"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.4.0", "firmwares/NINA/1.4.0/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(asList("Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.3.0", "firmwares/NINA/1.3.0/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010", "Arduino NANO 33 IoT"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.3.0", "firmwares/NINA/1.3.0/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(asList("Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.4", "firmwares/NINA/1.2.4/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010", "Arduino NANO 33 IoT"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.4", "firmwares/NINA/1.2.4/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(asList("Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.3", "firmwares/NINA/1.2.3/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.3", "firmwares/NINA/1.2.3/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(asList("Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.2", "firmwares/NINA/1.2.2/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.2", "firmwares/NINA/1.2.2/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(asList("Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.1", "firmwares/NINA/1.2.1/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.2.1", "firmwares/NINA/1.2.1/NINA_W102-Uno_WiFi_Rev2.bin", true, 1000000, new ArrayList<String>(asList("Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.1.0", "firmwares/NINA/1.1.0/NINA_W102.bin", true, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010", "Arduino MKR Vidor 4000", "Arduino Uno WiFi Rev2"))));
fwAvailable.add(new NinaFlasher("NINA firmware", "1.0.0", "firmwares/NINA/1.0.0/NINA_W102.bin", false, 1000000, new ArrayList<String>(asList("Arduino MKR WiFi 1010", "Arduino MKR Vidor 4000", "Arduino Uno WiFi Rev2"))));
```

p-brane, 2021-01-20:
Hi @per1234 
Thank you!
At the time I was using WiFiNINA 1.8.0 and it was using #define WIFI_FIRMWARE_LATEST_VERSION "1.4.2". I see that WiFiNINA 1.8.1 is back to #define WIFI_FIRMWARE_LATEST_VERSION "1.4.1"

closed 2025-03-20 by pennam