arduino/nina-fw #60: What's the main different between nina-fw and esp-at?

andy0808, 2020-12-10:
Hi guys : )

I'm new here. I plan to add an ESP32 as a slave device in my project. I'm wondering which firmware to choose.

Since there is already an `esp-at` project, and you can config the interface to SPI. So the speed is not a problem. Could I ask what the advantage `nina-fw` is compared to `esp-at`?

Thanks very much!

jandrassy, 2020-12-10:
AT firmware is controlled with AT commands usually over UART (Serial) (SPI is possible).
nina-fw is controlled with binary commands over SPI for WiFi and over UART for BLE.

nina-fw WiFi is controled from host MCU with WiFiNINA ibrary. BLE with ArduinoBLE library.
AT firmware is controlled from host directly with AT command or for WiFi my WiFiEspAT library can be used.

and it could be important that AT firmware can run only one TCP server

andy0808, 2020-12-10:
@jandrassy Wow, thank you for your rapid response!

So besides the communication/control differences, the main functional difference is **AT firmware can not run multiple TCP servers**.

Got it! I'll check it out if I have the requirement in my project. Thanks very much!

closed 2020-12-11 by andy0808