arduino/nina-fw #63: WiFiServer returns disconnected WiFiClients after WiFi reconnect

jridgewell, 2021-01-27:
After WiFi disconnects, a `WiFiServer` will return disconnected `WiFiClients` from `::available()`. I believe this is caused by the nina-fw not properly clearing out the `socketTypes` (and resetting disconnected clients/servers/UDPs) in [`CommandHandlerClass::handleWiFiDisconnect()`](https://github.com/arduino/nina-fw/blob/200f09d1d5aa867db1cb8b5691ba90ed83ca3bbe/main/CommandHandler.cpp#L1615-L1634).

I can only get this to trigger by creating both a `WiFiServer` and a `WiFiUDP`. But with this example, I'm able to reproduce it 100%. If I modify the WiFiNINA library to log the sockets, I can see that the eventual `WiFiClient` is created with socket `0`, which is the same one the original `WiFiServer` was bound with.

```c++
#include <WiFiNINA.h>
#include "arduino_secrets.h"

#define OTA_PORT 65280

const char SSID[] = SECRET_SSID;
const char PASS[] = SECRET_PASS;
bool ledState = LOW;
unsigned long start = 0;
bool reconnected = false;

WiFiServer server(OTA_PORT);
WiFiUDP mdnsSocket;

void setup() {
  // Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial);

  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    while (true);
  }

  pinMode(PIN_LED, OUTPUT);
  digitalWrite(PIN_LED, LOW);

  delay(1500);
  connectWiFi();
  connectArduinoOTA();
}

void loop() {
  // 5 seconds after booting, disconnect and reconnect WiFi
  if (start == 0) {
    start = millis();
  } else if (!reconnected && millis() - start > 2000) {
    Serial.println("Disconnecting from WiFi");
    WiFi.disconnect();
    delay(100);
    connectWiFi();
    connectArduinoOTA();
    reconnected = true;
  }

  ledState = !ledState;
  digitalWrite(PIN_LED, ledState);

  WiFiClient client = server.available();
  if (client) {
    Serial.println("Client found");

    if (!client.connected()) {
      Serial.println("But it wasn't connected!");
      while (true);
    }

    delay(10);
    client.stop();
  }

  delay(200);
}

void connectWiFi() {
  do {
    Serial.println("Attempting to connect to SSID: " + String(SSID));
  } while (WiFi.begin(SSID, PASS) != WL_CONNECTED);
}

void connectArduinoOTA() {
  server.begin();
  mdnsSocket.beginMulticast(IPAddress(224, 0, 0, 251), 5353);
}
```

```text
01:26:26.754 -> Attempting to connect to SSID: Test
01:26:29.830 -> WiFiServer connected with socket 0
01:26:29.830 -> WiFiUDP connected with socket 1
01:26:31.857 -> Disconnecting from WiFi

01:26:31.962 -> Attempting to connect to SSID: Test
01:26:32.176 -> WiFiServer connected with socket 2
01:26:32.176 -> WiFiUDP connected with socket 1
01:26:32.176 -> WiFiClient constructed with socket 0
01:26:32.176 -> Client found
01:26:32.176 -> But it wasn't connected!
```

ocrdu, 2021-01-30:
Haven't thought long and hard yet, but could this be releated to https://github.com/arduino-libraries/WiFiNINA/issues/87 ?

lcoffin, 2023-04-29:
Anyone have a fix for this? Or a workaround? I seem to be having the same trouble although I don't always get the client.connected() returning false -- some times that passes and the request from the remote machine comes in, but the reply doesn't appear to get back to the remote machine. In any case, this is happening after the WiFi disconnects and reconnects.

I've tried making the server variable dynamic -- creating a new instance each time the WiFi comes up, then deleting the old one when it disconnects, but that doesn't seem to help.

I don't see a proper WiFiServer::stop() method defined that might do a more proper cleanup/shut down of sockets.

Any thoughts on how to fix this?