arduino/nina-fw #65: client.status always returns 4 even with session is disconnected

Jauwaad, 2021-04-20:
https://github.com/arduino-libraries/WiFiNINA/issues/159

This issue is reproducible in version 1.4.3 but doesn't exist on 1.3.0

lasselukkari, 2021-04-20:
@Jauwaad, hello again! It's been a while. I'm sad to hear you are having issues with this hardware stack again. I debugged your previous problem as you ran into the problem while using my code. This time it's not even remotely related to any codebase I'm maintaining. 

My advise is to use some other hardware stack.

Jauwaad, 2021-04-20:
@lasselukkari Thanks for the info. Just keeping you in the loop as you had previously helped. Unfortunately we're tied to the hardware stack.

lasselukkari, 2021-04-21:
Maybe you could run the Arduino ESP32 core on the u-blox module with your custom code and then just communicate over serial with the ARM microcontroller. This way you would have more control over how things work and you would not be stuck with this.

Jauwaad, 2021-04-21:
Unfortunately its too late to change platforms. Our primary requirement is USB Host hardware

aentinger, 2021-05-25:
Should be fixed with https://github.com/arduino/nina-fw/releases/tag/1.4.5.

closed 2021-05-25 by aentinger