arduino/nina-fw #75: WifiNiNA cannot handle when the same SSID is broadcasted from multiple APs

borianson, 2021-09-09:
We tried multiple MKR1010 boards with multiple different WifiNiNA FW revisions but could not establish a stable connection via wifi (or at the wast majority of the cases even connect). We have a grid of ~20 pcs of well working brand new Ubiquiti UniFi APs broadcasting the same SSID at our facility. Based on other users experience at the Arduino forum and on our experiance we assume something is wrong with how the FW or the module itself handles this specific situation when the device can detect multiple APs broadcasting the same SSID.

Russell108, 2022-11-24:
Yes I have the same issues
We have a network of Access points Unifi I have found even if i get it to connect to the nearest Access point the Arduino MKR 1010 or nano IOT 33 can move to another over time. Usually one with a ridiculously low signal.

CptHolzschnauz, 2023-05-10:
I confirm. I have the same SSID on a 2.4 and 5 GHz band. MKR1010 can't handle the setup. WifiNina Firmware 1.5 WPA2 Personal.

g6094199, 2023-11-07:
+1 

with mkr1010 an fw 1.5

and a bunch of freifunk/openwrt APs (atheros ath10k based)

on the mkr it looks like this:

```
I am connected to the network
SSID: 
BSSID: 00:00:00:00:00:00
signal strength (RSSI): 0 dBm
Encryption Type:7

My IP Address: 192.168.xx.114
My MAC address: 30:AE:xx:xx:xx:04
```

no SSID from WiFi.SSID() and only zeros from WiFi.BSSID(), rssi = 0

tmcqueen-materials, 2024-04-23:
@per1234 : This seems to be quite a fatal flaw in the current nina-fw, as it makes completely useless boards such as the Nano 33 IoT at home with a mesh network. Since it sounds like the solution is "easy" (change back to WIFI_ALL_CHANNEL_SCAN), can that be done with a small iteration of the firmware so this doesn't break for a situation that is just becoming more common? Thanks!

nhbn, 2024-09-09:
We also have a multiple AP setup (Ubiquiti). We wanted to use Nano 33 IoT and I have about 20 Arduino sitting in the shelf to make them work. At the moment, we are struggeling with unstable connections (due to this issue).

Being able to use WIFI_ALL_CHANNEL_SCAN could be an option, being able to connect using BSSID would be optimal.

I'm considering switching to ESP32 that allow for that behaviour (though we would prefer the arduino eco system).

Is there a reason why this issue is open for so long?

Russell108, 2024-09-09:
My understanding is Arduino team need to fix it 
or someone needs to offer a patch I guess.

I used to ask   but the Arduino team made it very clear they felt it inappropriate that I ask
and would not provide any updates and threatened to report me to git hub if I continued to ask for updates.

Indeed An ESP 32 might be the way to do if it suits. I still use the Arduino

My solution is to create an SSID on the network and only have it on one Access point.
I have the benefit of running the network

borianson, 2024-09-10:
In the mean time we migrated to ESP32 and I suggest you the same. It is a pity that this could not be retified during the many years passed.

> We also have a multiple AP setup (Ubiquiti). We wanted to use Nano 33 IoT and I have about 20 Arduino sitting in the shelf to make them work. At the moment, we are struggeling with unstable [...]

In the mean time we migrated to ESP32 and I suggest you the same. It is a pity that this could not be retified during the many years passed.

tgomer, 2025-03-06:
This means that IOT projects using WifiNiNa should be cancelled cause the risk of failing for different network environments.
I fell into that trap also and am migrating to ESP32 too. How about to warn the community? This issue is not obvious at all.

wasteofpaint, 2025-04-16:
Is there any reason why it's set to WIFI_FAST_SCAN instead of WIFI_ALL_CHANNEL_SCAN?

As far as I understand it, this fix is a quick one line change in the arduino/libraries/wifi/WiFi.cpp file. 

I really don't understand breaking multi AP compatibility to save a second or two on connections. Multi AP setups are VERRY common these days. 

Any chance this could be changed in the upcoming 3.00 FW release?? Please?

g6094199, 2025-04-16:
@wasteofpaint 

thats a VERY GOOD FIND! thank you. 
i will try/test it tomorrow. my setup is a hostpital with dozens of ap's  with several vap's per unit where i measure air quality.

wasteofpaint, 2025-04-16:
@g6094199 

Great! Let us know how it works for you. 

It's kind of annoying to have to download the toolchain and re-build the firmware when this really should be the default behavior - I haven't done this yet myself.  

I'm still hopeful that it will be changed in the official 3.0.0 release though.

wasteofpaint, 2025-04-17:
See: 
https://github.com/arduino/nina-fw/pull/9

g6094199, 2025-04-29:
> WIFI_FAST_SCAN

is no valid system call on wifinina. there are major differences to the esp32 wifi driver.

wasteofpaint, 2025-04-29:
@g6094199 

You can't set  WIFI_ALL_CHANNEL_SCAN though the wifinina library. You need to change it in the NINA FW wifi.cpp file, rebuild the fw, and flash it to the wifi module. 

I've created a pull request with this change. Hopefully they will choose to merge it into the official NINA FW. This would greatly simplify things.

esaunoya, 2026-09-02:
We hit this in production (three UniFi APs, one SSID) and can add hard numbers to the thread: at the moment of association our MKR WiFi 1010 could *hear* the nearest AP at **−61 dBm on ch 11**, yet fast scan joined a **−81 dBm AP on ch 6** — five consecutive reconnects, every time, because both weak APs sit on lower channels and `WIFI_FAST_SCAN` takes the first match in channel order. The weak associations stalled TLS long enough to trip our host watchdog repeatedly.

Also confirming @Russell108's workaround works well as a stopgap: an SSID broadcast from only one AP sidesteps the selection problem entirely.

Per the maintainer guidance in #118, I've implemented the opt-in version as a pair of draft PRs: arduino/nina-fw#131 (new `SET_SCAN_MODE` command, default unchanged) + arduino-libraries/WiFiNINA#332 (`WiFi.setScanMethod()`). Hardware validation on the same three-AP network is in progress.

closed 2026-10-29 by dani-clo