arduino/nina-fw #85: WiFi.status() returns 255 randomly but remains connected and operational

Roamer105, 2023-02-06:
A significant web server sketch running on an MKR1010 experiences random "disconnects" which are not really disconnects since the status returned is 255 and not any of the valid status codes listed. For now, simply ignoring it and treat it the same as WL_CONNECTED overcomes the issue as a work-around.

Roamer105, 2023-02-09:
Additional information: the web server simultaneously uses Wi-Fi to connect to a ModBus server as a client to collect information periodically. Disabling the client code appears to correct the problem. 
consequence:  it appears as that WFININA does not support both a web server and a client simultaneously.

dansiviter, 2024-05-10:
I'm seeing something very similar: https://forum.arduino.cc/t/wifi-connection-unreliable/1258323/2